ros-industrial/ros2_canopen #247: I think the document needs to be more optimized to facilitate understanding how to use Canopen

zouzhe1, 2024-01-12:
**I am a beginner**

I think the document needs to be more optimized to facilitate understanding how to use Canopen。

# A.words meaning
1，lifecycle
https://ros-industrial.github.io/ros2_canopen/manual/rolling/quickstart/examples.html
"cia402_lifecycle_setup.launch.py"  and  "cia402_setup.launch.py"，What is the difference between this “lifecycle”？
2，Proxy 
Why use the term proxy, which makes it difficult for people to immediately understand specific things

# B.user-guide
The user guide only contains scattered and messy knowledge introductions, which is very unclear and unfriendly to newcomers

# C.The most common example
If I only want a motor that supports Canopen cia402 to rotate in speed mode or position mode. What should I do? Isn't this the most common and commonly used situation? Looking at you again, you have provided a bunch of controllers, but none of them have clear information that can be used in this way.

# D.suggest
You should specifically write a page on **how to control the rotation of a cia402 servo**.

# E.about controller name
The controller should have a name that is understandable, such as a **position_mode_servo_controller** or a **speed_mode_servo_controller**

Let's take a look at the names you've given us, forward_ Position_ Controller, when the newcomer sees this name, they only know that it is related to the location and do not know if they need anything 

# F.Roast
https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-robot-system.html
I followed this page and it doesn't work at all because it's not detailed enough.

Finally, there is only one sentence testing cli, without any formal code on how to control it

Lars-Ki, 2024-01-31:
Hello,

I came across this issue while trying to integrate a motor that supports CANopen CiA 402 into a project using a single-board computer and a CAN module. My goal is to control the motor in either speed mode or position mode, which I believe is a common use case for many developers.

So far, I've established communication between the motor and the single-board computer, as evidenced by the motor driver responding with an error message. This leads me to believe that the wiring is correctly set up. However, I'm encountering difficulties in finding clear information or examples on how to proceed with the actual motor control using the ros2 canopen package. I have now spent multiple 8-hour shift days trying to get started.

A super simple example or a snippet of code demonstrating how to get a motor running in either speed or position mode using the ros2 canopen package would be immensely helpful. I'm particularly interested in any prerequisites, configuration steps, or specific commands required to achieve basic motor operation within this context.

Thank you in advance for any guidance or resources you can provide. It would be greatly appreciated by those of us trying to navigate these functionalities for our projects.

Best regards,

Lars-Ki

tejasps28, 2024-04-17:
@Lars-Ki hi, were you able to implement controlling motor using this ros2_canopen?

roncapat, 2024-04-17:
@zouzhe1 fully agree with you. BTW, have you understood how to activate a cia_402 drive in position or velocity?

zouzhe1, 2024-07-24:
@roncapat Of course, I know how to make the motor start in speed and position mode. Simply put, it means resetting the NMT first when powered on, and then starting. Then, let 6040 go from 0-6-7-15
But this ros2canopen library is simply not user-friendly. I wonder if it has been a long time and is now working? Their documentation is very poor! And surprisingly, he didn't use a few commonly used simple answer routines.

hellantos, 2024-07-24:
Hi guys, sorry changed jobs and moved... here you might find a more detailed tutorial on how to create your own robot. Please be aware that this is a beta version library. https://hellantos.github.io/robby_description/

zouzhe1, 2024-07-25:
> Hi guys, sorry changed jobs and moved... here you might find a more detailed tutorial on how to create your own robot. Please be aware that this is a beta version library. https://hellantos.github.io/robby_description/
> 
> Also to use it in a meaningful way you'll need to know how to model a robot with urdf an ros2_control.

@hellantos 
Thank you very much for your response. Your link is very useful. Very clear, especially with regards to the pdo configuration. What makes me feel difficult is how Xacro learns? There is a feeling of looking at front-end HTML CSS. Does Ros2 control require a controller? Do we have to write our own controllers if we don't have the ones provided by the official authorities? Is it the essence of Ros2 control or does the external program communicate with Ros2 control through topics and so on? If one of the motors in CIA402 malfunctions, will the ros2 canopen package automatically handle the process of CIA402 and make it work again on 6-7-15?

hellantos, 2024-07-25:
@zouzhe1
There is the question what your use case is. If you just want à drive at constant speed and that's it, then probably this lib is a bit overkill.

If you want to build a robot arm or mobile one with multiple drives that need to work precisely together, then this library and ros2_control will probably be the solution to go with.

About xacro/urdf there are I believe some good instructions in the construct's courses.