ros-industrial/ros2_canopen #248: Build Failure on ROS 2 Buildfarm for Rolling

Yadunund, 2024-01-13:
See
- https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__canopen_ros2_controllers__ubuntu_jammy_amd64__binary/

Full Log from [Console Output](https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__canopen_ros2_controllers__ubuntu_jammy_amd64__binary/100/console) :

<details>
  <summary>Click me</summary>
  
```bash
15:07:53 In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:387,
15:08:01                  from /opt/ros/rolling/src/gmock_vendor/include/gmock/internal/gmock-internal-utils.h:47,
15:08:01                  from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock-actions.h:145,
15:08:01                  from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock.h:59,
15:08:01                  from /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/test/test_canopen_proxy_controller.hpp:27,
15:08:01                  from /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/test/test_canopen_proxy_controller.cpp:15:
15:08:01 /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/test/test_canopen_proxy_controller.hpp: In member function ‘void CanopenProxyControllerFixture<CtrlType>::SetUpController(bool, std::string)’:
15:08:01 /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/test/test_canopen_proxy_controller.hpp:133:32: error: no matching function for call to ‘TestableCanopenProxyController::init(std::string&)’
15:08:01   133 |     ASSERT_EQ(controller_->init(controller_name), controller_interface::return_type::OK);
15:08:01       |               ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
15:08:01 In file included from /opt/ros/rolling/include/controller_interface/controller_interface/controller_interface.hpp:22,
15:08:01                  from /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/include/canopen_ros2_controllers/canopen_proxy_controller.hpp:26,
15:08:01                  from /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/test/test_canopen_proxy_controller.hpp:26,
15:08:01                  from /tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/test/test_canopen_proxy_controller.cpp:15:
15:08:01 /opt/ros/rolling/include/controller_interface/controller_interface/controller_interface_base.hpp:116:23: note: candidate: ‘virtual controller_interface::return_type controller_interface::ControllerInterfaceBase::init(const string&, const string&, unsigned int, const string&, const rclcpp::NodeOptions&)’
15:08:01   116 |   virtual return_type init(
15:08:01       |                       ^~~~
15:08:01 /opt/ros/rolling/include/controller_interface/controller_interface/controller_interface_base.hpp:116:23: note:   candidate expects 5 arguments, 1 provided
15:08:01 make[4]: *** [CMakeFiles/test_canopen_proxy_controller.dir/build.make:79: CMakeFiles/test_canopen_proxy_controller.dir/test/test_canopen_proxy_controller.cpp.o] Error 1
15:08:01 make[4]: Leaving directory '/tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/.obj-x86_64-linux-gnu'
15:08:01 make[3]: *** [CMakeFiles/Makefile2:294: CMakeFiles/test_canopen_proxy_controller.dir/all] Error 2
15:08:01 make[3]: Leaving directory '/tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/.obj-x86_64-linux-gnu'
15:08:01 make[2]: *** [Makefile:149: all] Error 2
15:08:01 make[2]: Leaving directory '/tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7/.obj-x86_64-linux-gnu'
15:08:01 dh_auto_build: error: cd .obj-x86_64-linux-gnu && make -j1 VERBOSE=1 returned exit code 2
15:08:01 make[1]: Leaving directory '/tmp/binarydeb/ros-rolling-canopen-ros2-controllers-0.2.7'
15:08:01 make[1]: *** [debian/rules:44: override_dh_auto_build] Error 25
15:08:01 make: *** [debian/rules:27: build] Error 2
15:08:01 dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
15:08:01 Traceback (most recent call last):
15:08:01   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 149, in build_binarydeb
15:08:01     subprocess.check_call(cmd, cwd=source_dir, env=env)
15:08:01   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
15:08:01     raise CalledProcessError(retcode, cmd)
15:08:01 subprocess.CalledProcessError: Command '['dpkg-buildpackage', '-b', '-us', '-uc']' returned non-zero exit status 2.
```
</details>

Rolling sync is blocked as a result of this regression.

hellantos, 2024-01-19:
@Yadunund have tried to push release the changes, but bloom fails when trying to create the PR

```
Open a pull request from 'ipa-cmh/rosdistro:bloom-ros2_canopen-1' into 'ros/rosdistro:master'?
Continue [Y/n]? Y
==> git checkout -b bloom-ros2_canopen-1
Switched to a new branch 'bloom-ros2_canopen-1'
==> Pulling latest rosdistro branch
remote: Enumerating objects: 196047, done.
remote: Counting objects: 100% (55112/55112), done.
remote: Compressing objects: 100% (182/182), done.
remote: Total 196047 (delta 54910), reused 54957 (delta 54904), pack-reused 140935
Receiving objects: 100% (196047/196047), 94.48 MiB | 15.03 MiB/s, done.
Resolving deltas: 100% (115755/115755), done.
From https://github.com/ros/rosdistro
 * branch                master     -> FETCH_HEAD
==> git reset --hard 3b79ff532181e44971eb7a4fe6efdb0566b990d7
HEAD is now at 3b79ff532 tiago_simulation: 4.0.8-1 in 'humble/distribution.yaml' [bloom] (#39587)
==> Writing new distribution file: rolling/distribution.yaml
==> git add rolling/distribution.yaml
==> git commit -m "ros2_canopen: 0.2.8-1 in 'rolling/distribution.yaml' [bloom]"
[bloom-ros2_canopen-1 771574dbf] ros2_canopen: 0.2.8-1 in 'rolling/distribution.yaml' [bloom]
 1 file changed, 1 insertion(+), 1 deletion(-)
==> Pushing changes to fork
Enumerating objects: 7313, done.
Counting objects: 100% (7313/7313), done.
Delta compression using up to 8 threads
Compressing objects: 100% (1863/1863), done.
Writing objects: 100% (7281/7281), 8.67 MiB | 8.08 MiB/s, done.
Total 7281 (delta 3759), reused 7214 (delta 3696), pack-reused 0
remote: Resolving deltas: 100% (3759/3759), completed with 23 local objects.
remote: This repository moved. Please use the new location:
remote:   https://github.com/hellantos/rosdistro.git
remote: 
remote: Create a pull request for 'bloom-ros2_canopen-1' on GitHub by visiting:
remote:      https://github.com/hellantos/rosdistro/pull/new/bloom-ros2_canopen-1
remote: 
To https://github.com/ipa-cmh/rosdistro.git
 * [new branch]          bloom-ros2_canopen-1 -> bloom-ros2_canopen-1
Failed to open pull request: GithubException - HTTP Error 422: Unprocessable Entity (https://api.github.com/repos/ros/rosdistro/pulls): None
```

Not sure why...

Yadunund, 2024-01-19:
I'm not sure what the cause of the failure is but that last step in bloom which opens the PR on rosdistro can be manually created. So you can manually branch off the latest `master` of `rosdistro`, bump the version of your repo in `iron/distribution.yaml` and open a PR.