ros-industrial/ros2_canopen #251: HOW to build ros2 canopen raspberrypi4B  on x86 platform with cross-compilation?

zouzhe1, 2024-01-20:
Target Device: Raspberry Pi 4B with Ubuntu 22,
 Local Device: x86_64 with Ubuntu 22. 

Due to the limited memory on my Raspberry Pi 4B, the compilation speed of ROS 2 CANopen is extremely slow, and it often freezes. 

Therefore, I intend to perform cross-compilation on my PC. 

How should I proceed with cross-compilation?

 It would be preferable if you could provide a binary version that can be installed using sudo.

LudoJ49, 2024-01-25:
use this to create 2G of extra swap memory...

    sudo fallocate -l 2G /swapfile    # Adjust size as needed (2G in this example)
    sudo chmod 600 /swapfile
    sudo mkswap /swapfile
    sudo swapon /swapfile

MichaelT800, 2024-05-13:
Since lely_core_libraries is generated for x86 architecture, you need to modify the CMakeLists.txt of this package.
# add:Cross-compilation or normal compilation
if(CMAKE_CROSSCOMPILING)
  set(CONFIGURE_COMMAND_PREFIX . ${CMAKE_CURRENT_SOURCE_DIR}/arm64_toolchain.sh &&)
  set(CONFIGURE_HOST_OPTION --host=aarch64-linux-gnu)
else()
  set(CONFIGURE_COMMAND_PREFIX "")
  set(CONFIGURE_HOST_OPTION "")
endif()
# modifications
  CONFIGURE_COMMAND ${CONFIGURE_COMMAND_PREFIX} autoreconf -i <SOURCE_DIR> && <SOURCE_DIR>/configure ${CONFIGURE_HOST_OPTION} --prefix=<INSTALL_DIR> --disable-cython --disable-doc --disable-tests --disable-static --disable-diag
as such
[image]