ros-industrial/ros2_canopen #255: basic build error with preprocessed_bus.yml  ???

LudoJ49, 2024-01-27:
Hi,
I'm tyrying to create a master following the datasheet. 
I have done yml and launch file following the doc

[image]

I have the below error. Why is it looking for preprocessed_bus.yml nit bus.yml ?

**logs:**
<pre>Starting &gt;&gt;&gt; my_canopen

--- stderr: my_canopen                         

sed: can&apos;t read /home/ludo/build/my_canopen/config/myCanOpenBus/preprocessed_bus.yml: No such file or directory

gmake[2]: *** [CMakeFiles/myCanOpenBus.dir/build.make:74: myCanOpenBus] Error 2

gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/myCanOpenBus.dir/all] Error 2

gmake: *** [Makefile:146: all] Error 2

---

Failed   &lt;&lt;&lt; my_canopen [1.91s, exited with code 2]

                                

Summary: 0 packages finished [2.81s]

  1 package failed: my_canopen

  1 package had stderr output: my_canopen

</pre>

**Setup:**
 - Device: virtual machine
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: Humble

ipa-vsp, 2024-01-29:
Hey @LudoJ49, 

it seems like there's an issue with your 'bus.yml' file. Could you please share the contents of that file? 

The preprocessed yml file is generated from the `bus.yml`.

LudoJ49, 2024-01-30:
after renaming my bus directory, it seems to woek.

But I have another issue.

Where is that 1F85 object comming from ? I don't have it in my ESD file.

it seems the device does not go into operating stage...

<pre>
[device_container_node-1] [WARN] [1706640803.740556615] [my_canopen_bus]: Wait for device to boot.
[device_container_node-1] [ERROR] [1706640804.761704919] [my_canopen_bus]: Boot Issue: Value of object 1018:01 from CANopen device is different to value in object 1F85 (Vendor-ID).
[device_container_node-1] [INFO] [1706640804.761780391] [my_canopen_bus]: Driver booted and ready.
[device_container_node-1] [INFO] [1706640804.764332542] [my_canopen_bus]: Starting with event mode.
[device_container_node-1] sync_sdo_read_typed: id=4 index=0x6061 subindex=0 timed out.
</pre>

hellantos, 2024-07-24:
This is because the object 1018 in eds and objct 1018 of your device on the bus are not the same.