ros-industrial/ros2_canopen #262: Master and Proxy Node Namespaces

kurtist123, 2024-03-05:
Is it possible to add namespaces to the master and proxy CAN nodes? I have tried a few ways of getting the master and proxy nodes pushed into a namespace without success.

I am running on Ubuntu 22.04 using ROS2 Humble and using the latest humble branch in this repository.

I have attached my configuration files:
[bus.txt](https://github.com/ros-industrial/ros2_canopen/files/14497740/bus.txt)
[device_container_launch.txt](https://github.com/ros-industrial/ros2_canopen/files/14497727/device_container_launch.txt)
[node_list.txt](https://github.com/ros-industrial/ros2_canopen/files/14497728/node_list.txt)

- bus.txt: my bus.yml configuration starting a master node and a proxy node named can0_comm.
- device_container_launch.txt: a snippet of my launch file where I launch the device container.
- node_list.txt: output from `ros2 node list` showing which nodes are namespaced and which ones are not.

In my launch file (device_container_launch.txt) I namespace the device container with "dummy_ns". I was expecting this namespace to be propagated to the master and proxy nodes as well.

As you can see in node_list.txt, the device_container node has the correct namespace but the master and can0_comm proxy nodes are not namespaced. Is it possible to move the master and proxy nodes into the "dummy_ns" namespace as well?

I have also tried pushing the device_container into a namespace using launch_ros.actions.PushRosNamespace and get the same results as above.

My use case is running multiple robots in simulation. If each robot has a master and proxy node that is not namespaced I will not be able to run multiple robots in simulation on the same network at the same time.

ipa-vsp, 2024-04-23:
Hi @kurtist123,
PR #270 should introduce namespacing. Additionally, the humble branch has been synced with the latest changes, but I haven't had a chance to test it yet.

kurtist123, 2024-09-13:
This has been solved. Closing this issue. Thanks.

closed 2024-09-13 by kurtist123