ros-industrial/ros2_canopen #268: Please help me see what's going on?

leensoft, 2024-03-19:
##### Service calls producing errors:

```
ros2 service call /cia402_device_2/init std_srvs/srv/Trigger
```

```

[device_container_node-1] sync_sdo_read_typed: id=2 index=0x6502 subindex=0 timed out.
[device_container_node-1] sync_sdo_read_typed: id=2 index=0x6502 subindex=0 timed out.
[device_container_node-1] [ERROR] [1710836525.064384496] [cia402_device_2]: AsyncUpload:02:6502:00: Object does not exist in the object dictionary (06020000): Object does not exist in the object dictionary
[device_container_node-1] [INFO] [1710836525.064486575] [canopen_402_driver]: Init: Read State
[device_container_node-1] [INFO] [1710836525.064531196] [canopen_402_driver]: Init: Enable
[device_container_node-1] [INFO] [1710836528.553105211] [canopen_402_driver]: Fault reset
[device_container_node-1] [INFO] [1710836530.065283573] [canopen_402_driver]: Transition timed out.
[device_container_node-1] Could not enable motor
```

##### Object definition in object dictionary:

```
[6502]
ParameterName=Supported Drive Modes 1
ObjectType=0x07
DataType=0x0007
AccessType=ro
DefaultValue=0x0000002D
PDOMapping=1
```

##### Bus config:

```
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 10
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  baud_rate: 250
  heartbeat_consumer: false
  heartbeat_producer: 2000
  start_nodes: true
  sync_period: 5000000 
 
defaults:
  dcf: "Kinco_EDS_20210508.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  period: 20
  polling: false
  heartbeat_producer: 1000
  # switching_state: 2
  # position_mode: 1
  scale_pos_from_dev: 0.03
  scale_pos_to_dev: 31.4
  sdo:
    - {index: 0x60C2, sub_index: 1, value: 50} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
    - {index: 0x6081, sub_index: 0, value: 1000}
    - {index: 0x6083, sub_index: 0, value: 2000}
  tpdo: # TPDO needed statusword, actual velocity, actual position, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6061, sub_index: 0} # mode of operation display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6064, sub_index: 0} # position actual value
        - {index: 0x606c, sub_index: 0} # velocity actual position
    3:
      enabled: false
    4:
      enabled: false
  rpdo: # RPDO needed controlword, target position, target velocity, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0} # controlword
      - {index: 0x6060, sub_index: 0} # mode of operation
    2:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x607A, sub_index: 0} # target position
      - {index: 0x60FF, sub_index: 0} # target velocity

nodes:
  # cia402_device_1:
  #   node_id: 1 
  cia402_device_2:
    node_id: 2 

```

**Setup:**

 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: humble

glaciercoder, 2024-03-26:
I don't understant why it logged error about 0602. It seems you didn't use 0602.

glaciercoder, 2024-03-26:
Have you solved the problem? I'm also using Kinco device. @leensoft

leensoft, 2024-03-28:
> Have you solved the problem? I'm also using Kinco device. @leensoft

The problem hasn't been solved, have you solved it?

nicklgw, 2024-03-28:
In EDS file, you can add a line about 0602 to fix it.

cschindlbeck, 2024-07-31:
Please provide a more descriptive Issue title