ros-industrial/ros2_canopen #321: Ubuntu 24.04 Support and road map missing ?

pallav-xlnx, 2024-09-18:
There are no guidelines or information regarding the compatibility with ROS2 Jazzy release. 

We are using Ubuntu 24.04 as our platform and ROS2 Jazzy is supported for Ubuntu noble (Note, humble is no more supported for Ubuntu noble). So for using ros2_canopen, we will need ROS2 Jazzy compatible branch/Tag. There are branches for humble, iron and other releases; is there a plan to have a stable tag or branch for ROS2-Jazzy.

Can someone please guide me, on what commit/tag/branch should be used on ubuntu 24.04? Will it be moving or stable enough to  be used? 

Also, is there any additional PPA for ros2_canopen, which can install latest 0.2.12 (assuming 0.2.12 is the latest release for Jazzy)?

Thanks in advance.

zimmerman-pliant, 2024-09-26:
I'm interested in a binary release too, but I've got `ros2_canopen` built and running with Jazzy on Noble. `master` branch, `#eb219eb`

eholandcmu, 2025-04-17:
Does anyone know if there is a plan towards Jazzy implementation? Is this abandoned?

zimmerman-pliant, 2025-04-21:
As I understand it, it was released for Jazzy in December:

https://github.com/[ros2-gbp/ros2_canopen-release](https://github.com/ros2-gbp/ros2_canopen-release?tab=readme-ov-file#ros2_canopen-jazzy---030-1)?tab=readme-ov-file#ros2_canopen-jazzy---030-1

I'm still using our fork because I need #316 for my application, so I haven't tested it from binary installation yet, but I've been using it on Jazzy successfully built in my workspace since shortly after my prior comment.