ros-industrial/ros2_canopen #332: How to pass parameter files to driver nodes

nsh-bgl, 2024-11-12:
I have created a new driver that inherits from the NodeCanopenProxyDriver. I would like to pass parameters to this new driver. 

I noticed that the CIA402 driver does so by using the bus.yml file. But I would like to pass a parameter file instead.

The reason for doing so is that the parameter file I am planning to use contains shared parameters that are used by more than one node.

I am launching my drivers using the device_container_node.

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: Humble

nsh-bgl, 2024-11-27:
I have been able to solve my issue by manually loading the YAML file locally and then extracting the parameter values.

for anyone interested:

`
#include "ament_index_cpp/get_package_share_directory.hpp"
std::string param_file = ament_index_cpp::get_package_share_directory("{Package/Name}") + "Path/to/config/file"

YAML::Node parameters = YAML::LoadFile(param_file);
`

closed 2024-11-27 by nsh-bgl

ZeroErrControl, 2025-01-25:
You can refer to [this project:https://github.com/ZeroErrControl/eRob_CANopen_Python](https://github.com/ZeroErrControl/eRob_CANopen_Python).