ros-industrial/ros2_canopen #338: Reported NMT state is incorrect

mrmitchadams, 2025-02-04:
**Describe the bug**
The reported NMT state is incorrect, at least from what I have observed while using the Proxy Driver

**To Reproduce**
Setup:
```
  executor_ = std::make_shared<rclcpp::executors::MultiThreadedExecutor>();
  device_container_ = std::make_shared<ros2_canopen::DeviceContainer>(executor_);
  executor_->add_node(device_container_);
  spin_thread_ = std::make_unique<std::thread>(&spin, this);
  device_container_->init(can_interface_name, master_config, bus_config);
  auto drivers = device_container_->get_registered_drivers();
  for (const auto& [driver_node_id, driver] : drivers)
  {
    auto proxy_driver = std::static_pointer_cast<ros2_canopen::ProxyDriver>(driver);
    proxy_drivers_[driver_node_id] = proxy_driver;
    auto nmt_state_cb = [this](canopen::NmtState nmt_state, uint8_t node_id) {
      std::cout << "NMT CB, State: " << static_cast<int>(nmt_state) << std::endl;
    };
   proxy_driver->register_nmt_state_cb(nmt_state_cb);
  }
```

I later start one of the drivers as follows:
```
proxy_drivers_.at(device_node_id)->start_node_nmt_command();
```
This results in:
- The NMT state of the driver, as observed via `candump`, transitioning to Operational
- The registered NMT state callback being called and printing out "NMT CB, state: 5"
- The following being logged: "Slave 0x3: Switched NMT state to START"
- An `/nmt_state` topic is published indicating a new NMT state of "START"

This is all as expected.

However, when I later run:
```
proxy_driver->reset_node_nmt_command();
```
- The NMT state of the driver, as observed via 'candump', transitions to Initializing, then to Pre-Operational, as expected
- The registered NMT state callback being called and printing out "NMT CB, state: 5"
- The following being logged: "Slave 0x3: Switched NMT state to START"
- An `/nmt_state` topic is published indicating a new NMT state of "START"

**Expected behavior**
calling `reset_node_nmt_command()` resets the drive to pre-operational mode, as expected. However, the feedback that is given through the NMT callback, the logs, and the ROS topic all indicate that the NMT state is switched to "START" (or Operational) rather than Pre-Operational.

**Logs**
```
[ros2_control_node-1] Reset NMT
[ros2_control_node-1] NMT CB, State: 5
[ros2_control_node-1] [INFO] [1738690767.237019553] [transport_base]: Slave 0x3: Switched NMT state to START
[ros2_control_node-1] Start NMT
[ros2_control_node-1] NMT CB, State: 5
[ros2_control_node-1] [INFO] [1738690772.048130197] [transport_base]: Slave 0x3: Switched NMT state to START
[ros2_control_node-1] Reset NMT
[ros2_control_node-1] NMT CB, State: 5
[ros2_control_node-1] [INFO] [1738690775.697599678] [transport_base]: Slave 0x3: Switched NMT state to START
[ros2_control_node-1] Start NMT
[ros2_control_node-1] NMT CB, State: 5
[ros2_control_node-1] [INFO] [1738690780.508815138] [transport_base]: Slave 0x3: Switched NMT state to START
[ros2_control_node-1] Reset NMT
[ros2_control_node-1] NMT CB, State: 5
[ros2_control_node-1] [INFO] [1738690784.157194190] [transport_base]: Slave 0x3: Switched NMT state to START
```

From candump:
```
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  00
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  00
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  05
  can0  703   [1]  00
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
  can0  703   [1]  7F
```

From ROS2 CLI:
```
# ros2 topic echo /drive/nmt_state 
data: START
---
data: START
---
data: START
---
data: START
---
data: START
---
```

**Setup:**
 - Device: Dell Precision 7670
 - OS: Ubuntu 22.04.5 LTS (Docker Container)
 - ROS-Distro: Humble
 - Branch/Commit: humble

viraatsworld, 2025-08-14:
I think the issue might be coming from :-
`  canopen::NmtState st = state;
  // We assume 1F80 bit 2 is false. All slaves are put into Operational after boot-up.
  // Lelycore does not track NMT states in this mode except BOOTUP.
  if (st == canopen::NmtState::BOOTUP)
  {
    st = canopen::NmtState::START;
  }`

in the file `lely_driver_bridge.cpp` inside the `canopen_base_driver` . I am not sure if removing this statement breaks the code but its worth a try.

viraatsworld, 2025-08-14:
I think the issue might be coming from :-

```cpp  
canopen::NmtState st = state;
  // We assume 1F80 bit 2 is false. All slaves are put into Operational after boot-up.
  // Lelycore does not track NMT states in this mode except BOOTUP.
  if (st == canopen::NmtState::BOOTUP)
  {
    st = canopen::NmtState::START;
  }
```

in the file `lely_driver_bridge.cpp` inside the `canopen_base_driver` . I am not sure if removing this statement breaks the code but its worth a try.