ros-industrial/ros2_canopen #342: 'Wait on device to boot' hang: Inconsistant booting of devices

mcdp-ppa, 2025-03-01:
**Issue** 
The entire stack compiles and launches normally, seems to be working. With one controller specified in the bus.yml it works almost all the time, with two controllers specified in the bus.yml the second one gets blocked on 'wait for device to boot' maybe 50% of the time, with three controllers the third gets blocked on 'wait for device to boot' maybe 90% of the time. I have 5 devices on the chain and I have never had more than 4 work since one eventually gets stuck at 'wait for device to boot' 
  

**To Reproduce** 

Here is the bus.yml: 
```
options:
  dcf_path: "@BUS_CONFIG_PATH@"

master:
  node_id: 1
  baudrate: 1000
  driver: "ros2_canopen::MasterDriver"
  package: "canopen_master_driver"
  sync_period: 20000
  boot_time: 10000
  boot_timeout: 10000

defaults:
  dcf: "AKD.eds"
  driver: "ros2_canopen::Cia402Driver"
  package: "canopen_402_driver"
  polling: false
  heartbeat_producer: 1000
  revision_number: 0x002C0000
  sdo:
    - {index: 0x60C2, sub_index: 1, value: 32} # Set interpolation time for cyclic modes to 50 ms
    - {index: 0x60C2, sub_index: 2, value: -3} # Set base 10-3s
  tpdo: # TPDO needed statusword, actual velocity, actual position, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6061, sub_index: 0} # mode of operation display
    2:
      enabled: true
      cob_id: "auto"
      transmission: 0x01
      mapping:
        - {index: 0x6041, sub_index: 0} # status word
        - {index: 0x6064, sub_index: 0} # position actual value
    3:
      enabled: false
    4:
      enabled: false
  rpdo: # RPDO needed controlword, target position, target velocity, mode of operation
    1:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0} # controlword
      - {index: 0x6060, sub_index: 0} # mode of operation
    2:
      enabled: true
      cob_id: "auto"
      mapping:
      - {index: 0x6040, sub_index: 0} # controlword
      - {index: 0x607A, sub_index: 0} # target position
    3:
      enabled: false
    4:
      enabled: false

nodes:
  joint_1:
    node_id: 2
[... 258 lines cut ...]
[device_container_node-1] Found tpdo mapped object: index=6041 subindex=0 
[device_container_node-1] Found tpdo mapped object: index=606c subindex=0 
[device_container_node-1] [WARN] [1740784992.819174401] [joint_4]: Wait for device to boot. 
[device_container_node-1] [INFO] [1740784993.660490380] [joint_3]: Slave 0x4: Switched NMT state to PREOP 
[device_container_node-1] [INFO] [1740784993.661130578] [joint_1]: Slave 0x2: Switched NMT state to PREOP 
[device_container_node-1] [INFO] [1740784993.661535780] [joint_2]: Slave 0x3: Switched NMT state to PREOP 
``` 

I have also changed the baud from 125k to 1M and nothing seems to change this behavior.  The only difference between a successful boot and an unsuccessful one is the wait for device to boot at the end.

**Setup:** 

- Device: Kollmorgen AKD 1207 
- OS: Ubuntu 22.04 using 5.15.0-1077-realtime kernel  
- ROS-Distro: Humble 
- Branch/Commit: Humble  
  
5x AKD1207 controllers connected through a CAN bus. These controllers have been setup to use CAN through the AKD CANopen Manual (which can be found [here](https://www.kollmorgen.com/sites/default/files/public_downloads/AKD%20CANopen%20Communications%20Manual%20EN%20%28REV%20U%29.pdf)). The CAN bus is connected by a FYSETC UCAN USB CAN adapter (the wiki is [here](https://wiki.fysetc.com/UCAN/) and the github with the hardware is [here](https://github.com/FYSETC/UCAN) through a USB port. I enable this via the following commands: 

``` 
sudo modprobe gs_usb  
sudo ip link set can0 up type can bitrate 1000000  
sudo ip link set can0 txqueuelen 1000  
sudo ip link set up can0  
```

islam120990, 2025-03-02:
Can you share the candump data log for both successful and unsuccessful case? We are also facing similar issue.

mcdp-ppa, 2025-03-10:
Hello! I have a workaround that is commenting out line 680 of lely_driver_bridge.hpp in canopen_base_driver

```
    //boot_cond.wait(lck);
```

Basically this is what causes the lock out which implies that `[lck](std::unique_lock<std::mutex> lck(boot_mtex))` is never released for some reason. I'm not sure what the downstream effects of this are but it now boots every time.

ipa-vsp, 2025-05-12:
Hi @mcdp-ppa ,

Apologies for the lack of support over the past few weeks, and thank you for your detailed report and follow-up.

PR #350 will serve as a workaround for this issue in the short term. We'll also investigate further to ensure the boot condition is notified properly in all cases.

I’ll go ahead and close this issue for now, but please feel free to reopen it if you encounter further problems or have any additional insights.

closed 2025-05-12 by ipa-vsp