ros-industrial/ros2_canopen #356: N5-2-2 motor controller timed out

javijurado, 2025-05-22:
**Describe the bug**
I'm trying to communicate with a N5-2-2 motor controller designed by Nanotec, but when I launch my master ant the driver for the motor controller I got an timed out
**Logs**
Output from the terminal:
```bash
ros2 launch app manager.launch.py 
[INFO] [launch]: All log files can be found below /home/intermodalics/.ros/log/2025-05-22-14-48-13-203659-im-laptop-016-1527
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: Launching canopen_core with output=screen
[INFO] [launch.user]: /home/javier/workspace/BPPAS/bppas-cleaning/install/share/can_manager/config/bus.yml
[INFO] [launch.user]: /home/javier/workspace/BPPAS/bppas-cleaning/install/share/can_manager/config/master.dcf
[INFO] [launch.user]: 
[INFO] [launch.user]: can0
[INFO] [device_container_node-1]: process started with pid [1543]
[device_container_node-1] [INFO] [1747918093.357806350] [device_container_node]: Starting Device Container with:
[device_container_node-1] [INFO] [1747918093.357886256] [device_container_node]:         master_config /home/javier/workspace/BPPAS/bppas-cleaning/install/share/can_manager/config/master.dcf
[device_container_node-1] [INFO] [1747918093.357896423] [device_container_node]:         bus_config /home/javier/workspace/BPPAS/bppas-cleaning/install/share/can_manager/config/bus.yml
[device_container_node-1] [INFO] [1747918093.357903036] [device_container_node]:         can_interface_name can0
[device_container_node-1] [INFO] [1747918093.358392610] [device_container_node]: Loading Master Configuration.
[device_container_node-1] [INFO] [1747918093.358636004] [device_container_node]: Load Library: /opt/ros/jazzy/lib/libmaster_driver.so
[device_container_node-1] [INFO] [1747918093.359986877] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1747918093.360005237] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::MasterDriver>
[device_container_node-1] [INFO] [1747918093.362511094] [master]: NodeCanopenBasicMaster
[device_container_node-1] [INFO] [1747918093.362522158] [device_container_node]: Load master component.
[device_container_node-1] [INFO] [1747918093.362528270] [device_container_node]: Added /master to executor
[device_container_node-1] [WARN] [1747918093.363013869] [master]: No timeout parameter found in config file. Using default value of 100ms.
[device_container_node-1] [INFO] [1747918093.363026136] [master]: Master boot timeout set to 2000ms.
[device_container_node-1] [INFO] [1747918093.370003298] [device_container_node]: Loading Driver Configuration.
[device_container_node-1] [INFO] [1747918093.370017545] [device_container_node]: Found device left_motor with driver ros2_canopen::Cia402Driver
[device_container_node-1] [INFO] [1747918093.370168344] [device_container_node]: Load Library: /home/javier/workspace/BPPAS/bppas-cleaning/install/lib/libcia402_driver.so
[device_container_node-1] [INFO] [1747918093.373300605] [device_container_node]: Found class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::Cia402Driver>
[device_container_node-1] [INFO] [1747918093.373325414] [device_container_node]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ros2_canopen::Cia402Driver>
[device_container_node-1] [INFO] [1747918093.376024774] [device_container_node]: Load driver component.
[device_container_node-1] [INFO] [1747918093.376052070] [device_container_node]: Added /left_motor to executor
[device_container_node-1] [INFO] [1747918093.382474645] [left_motor]: Non transmit timeout100ms
[device_container_node-1] [WARN] [1747918093.382560009] [left_motor]: Could not read enable diagnostics from config, setting to false.
[device_container_node-1] [INFO] [1747918093.382727946] [left_motor]: scale_pos_to_dev_ 1000.000000
[device_container_node-1] scale_pos_from_dev_ 0.001000
[device_container_node-1] scale_vel_to_dev_ 1000.000000
[device_container_node-1] scale_vel_from_dev_ 0.001000
[device_container_node-1] offset_pos_to_dev_ 0.000000
[device_container_node-1] offset_pos_from_dev_ 0.000000
[device_container_node-1] homing_timeout_seconds_ 10
[device_container_node-1] Found rpdo mapped object: index=6040 subindex=0
[device_container_node-1] Found rpdo mapped object: index=6060 subindex=0
[device_container_node-1] Found rpdo mapped object: index=3202 subindex=0
[device_container_node-1] Found rpdo mapped object: index=607a subindex=0
[device_container_node-1] Found rpdo mapped object: index=6081 subindex=0
[device_container_node-1] Found rpdo mapped object: index=6042 subindex=0
[device_container_node-1] Found rpdo mapped object: index=60fe subindex=1
[device_container_node-1] Found tpdo mapped object: index=6041 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6061 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6064 subindex=0
[device_container_node-1] Found tpdo mapped object: index=6044 subindex=0
[device_container_node-1] Found tpdo mapped object: index=60fd subindex=0
[device_container_node-1] [INFO] [1747918093.390671530] [left_motor]: Hey8 
[device_container_node-1] [INFO] [1747918093.390708568] [left_motor]: Hey5 
[device_container_node-1] [INFO] [1747918093.390746267] [left_motor]: Hey6 
[device_container_node-1] [INFO] [1747918093.390750010] [left_motor]: Hey7 
[device_container_node-1] [WARN] [1747918093.390758191] [left_motor]: Wait for device to boot.
[device_container_node-1] [INFO] [1747918093.390828463] [left_motor]: Driver booted and ready.
[device_container_node-1] [INFO] [1747918093.391053139] [left_motor]: Starting with event mode.
[device_container_node-1] [INFO] [1747918093.426445792] [left_motor]: Slave 0x1: Switched NMT state to START
[device_container_node-1] [INFO] [1747918093.669614870] [canopen_402_driver]: Fault reset
[device_container_node-1] sync_sdo_read_typed: id=1 index=0x606c subindex=0 timed out.
```

candump logs
```bash
  can0  703   [1]  00
[... 142 lines cut ...]
    return nodes_to_start

def generate_launch_description():
    declared_arguments = []

    declared_arguments.append(
        DeclareLaunchArgument(
            "can_interface_name",
            default_value="vcan0",
            description="CAN interface name to run the master and, when in simulation, the fake slaves.",
        )
    )
    
    return LaunchDescription(declared_arguments + [OpaqueFunction(function=launch_setup)])

```

**Setup:**
 - Device:
 - OS: Linux
 - ROS-Distro: Jazzy
 - Branch/Commit:

**Additional context**
Add any other context about the problem here.

mjimenezm00, 2026-01-09:
Hi! Thank you for your fast answer!

So, I already did that, as you can in the details of the can0:

```bash
ip -details link show can0
8: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/can 48:57:37:30:35:32:33:31:00:00:00:00:00:00:00:00 brd 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 promiscuity 0 minmtu 0 maxmtu 0 
    can <LOOPBACK> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 100 
	  bitrate 1000000 sample-point 0.750 
	  tq 83 prop-seg 4 phase-seg1 4 phase-seg2 3 sjw 1
	  ix_usb_can: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
	  clock 36000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus usb parentdev 3-6:1.0 
```

However, when running the ros2 launch canopen_core canopen.launch.py, it gets stuck in: Wait for device to boot.

Do you follow any specific sequence before launching?? I'm just currently setting up the can 0 with the baudrates and txqueuelen = 1000.

I'm running humble in commit: dc811bb6d2a1e3729f10c8581b863ac82ab6003c, as I saw this other issue: https://github.com/ros-industrial/ros2_canopen/issues/370

Thanks in advance :)

mjimenezm00, 2026-01-14:
Fixed it: https://github.com/ros-industrial/ros2_canopen/issues/370#issuecomment-3748813535