ros-industrial/ros2_canopen #359: Humble version does not support QoS

cecimaltese, 2025-06-06:
**Describe the bug**
when cloning the repo into ros2 Humble version ws this error appears:
--- stderr: canopen_core                                
/home/ceci/ros2_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp: In member function ‘void ros2_canopen::node_interfaces::NodeCanopenDriver<NODETYPE>::demand_set_master() [with NODETYPE = rclcpp::Node]’:
/home/ceci/ros2_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:30:32: error: cannot convert ‘rclcpp::QoS’ to ‘const rmw_qos_profile_t&’ {aka ‘const rmw_qos_profile_s&’}
   30 |     init_service_name, rclcpp::QoS(10), client_cbg_);
      |                                ^~~~~~~
      |                                |
      |                                rclcpp::QoS
In file included from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:1321,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/single_threaded_executor.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:22,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/ceci/ros2_ws/src/ros2_canopen/canopen_core/include/canopen_core/node_interfaces/node_canopen_driver.hpp:33,
                 from /home/ceci/ros2_ws/src/ros2_canopen/canopen_core/src/node_interfaces/node_canopen_driver.cpp:14:
/opt/ros/humble/include/rclcpp/rclcpp/node_impl.hpp:127:29: note:   initializing argument 2 of ‘typename rclcpp::Client<ServiceT>::SharedPtr rclcpp::Node::create_client(const string&, const rmw_qos_profile_t&, rclcpp::CallbackGroup::SharedPtr) [with ServiceT = canopen_interfaces::srv::CONode; typename rclcpp::Client<ServiceT>::SharedPtr = std::shared_ptr<rclcpp::Client<canopen_interfaces::srv::CONode> >; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_s; rclcpp::CallbackGroup::SharedPtr = std::shared_ptr<rclcpp::CallbackGroup>]’

**To Reproduce**
cd ~/ros2_ws/src
git clone https://github.com/ros-industrial/ros2_canopen.git
cd ..
rosdep install --from-paths src --ignore-src -r -y
colcon build
source install/setup.bash

**Setup:**
 - OS: Ubuntu 22.04
 - ROS-Distro: Humble
 - Branch/Commit: humble

rob9315, 2025-06-09:
please consider using the humble branch instead of the master branch for ROS2 Humble

```
cd ~/ros2_ws/src
git clone -b humble https://github.com/ros-industrial/ros2_canopen.git
cd ..
rosdep install --from-paths src --ignore-src -r -y
colcon build
source install/setup.bash
```

ipa-vsp, 2025-07-03:
@rob9315  Thank you for the answer! Now I have update the README for more clarity.