ros-industrial/ros2_canopen #370: Boot timeout PR #364 causing errors

jclinton830, 2025-07-23:
**Describe the bug**
A bu was introduced with PR #364. The CANOpen drivers no longer boot up and throw a boot timeout exception.

Reverting the this PR to this commit 52327f9b2286a83ef566304c8b0de5d2db177c02 fixes the issue. 

**To Reproduce**
Try to boot a canopen proxy driver with the new change released with that PR.

**Expected behavior**
Below is a snippet of the log

**Logs**
```
[ros2_control_node-3] [WARN] [1751933416.666535930] [fl_drive_joint]: Wait for device to boot...
[ros2_control_node-3] [ERROR] [1751933416.686735176] [fl_drive_joint]: Boot attempt 1 failed: Boot Timeout: The boot configure process timeout!
[ros2_control_node-3] [INFO] [1751933416.686836435] [fl_drive_joint]: Sending NMT reset before retrying boot
[ros2_control_node-3] [WARN] [1751933416.686867923] [fl_drive_joint]: Retrying boot configuration...
[ros2_control_node-3] [ERROR] [1751933416.707003152] [fl_drive_joint]: Boot attempt 2 failed: Boot Timeout: The boot configure process timeout!
[ros2_control_node-3] [INFO] [1751933416.707090490] [fl_drive_joint]: Sending NMT reset before retrying boot
[ros2_control_node-3] [WARN] [1751933416.707125305] [fl_drive_joint]: Retrying boot configuration...
[ros2_control_node-3] [ERROR] [1751933416.727276954] [fl_drive_joint]: Boot attempt 3 failed: Boot Timeout: The boot configure process timeout!
[ros2_control_node-3] [ERROR] [1751933416.727323864] [fl_drive_joint]: Boot failed after 3 attempts
[ros2_control_node-3] terminate called after throwing an instance of 'ros2_canopen::DriverException'
[ros2_control_node-3]   what():  std::exception
[ros2_control_node-3] Stack trace (most recent call last) in thread 14206:
[ros2_control_node-3] #21   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-3] #20   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in clone3 [0x7f905b9cbc3b]
[ros2_control_node-3] #19   Source "./nptl/pthread_create.c", line 447, in start_thread [0x7f905b93eaa3]
[ros2_control_node-3] #18   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x7f905bbcedb3, in 
[ros2_control_node-3] #17   Object "/ros_ws/install/groundhog_system/lib/libgroundhog_system.so", at 0x7f90217bff71, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (groundhog_system::GroundhogSystem::*)(), groundhog_system::GroundhogSystem*> > >::_M_run()
[ros2_control_node-3] #16   Object "/ros_ws/install/groundhog_system/lib/libgroundhog_system.so", at 0x7f90217c0609, in std::thread::_Invoker<std::tuple<void (groundhog_system::GroundhogSystem::*)(), groundhog_system::GroundhogSystem*> >::operator()()
[ros2_control_node-3] #15   Object "/ros_ws/install/groundhog_system/lib/libgroundhog_system.so", at 0x7f90217c158c, in void std::thread::_Invoker<std::tuple<void (groundhog_system::GroundhogSystem::*)(), groundhog_system::GroundhogSystem*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>)
[ros2_control_node-3] #14   Object "/ros_ws/install/groundhog_system/lib/libgroundhog_system.so", at 0x7f90217c1d1e, in std::__invoke_result<void (groundhog_system::GroundhogSystem::*)(), groundhog_system::GroundhogSystem*>::type std::__invoke<void (groundhog_system::GroundhogSystem::*)(), groundhog_system::GroundhogSystem*>(void (groundhog_system::GroundhogSystem::*&&)(), groundhog_system::GroundhogSystem*&&)
[ros2_control_node-3] #13   Object "/ros_ws/install/groundhog_system/lib/libgroundhog_system.so", at 0x7f90217c205a, in void std::__invoke_impl<void, void (groundhog_system::GroundhogSystem::*)(), groundhog_system::GroundhogSystem*>(std::__invoke_memfun_deref, void (groundhog_system::GroundhogSystem::*&&)(), groundhog_system::GroundhogSystem*&&)
[ros2_control_node-3] #12   Object "/ros_ws/install/groundhog_system/lib/libgroundhog_system.so", at 0x7f90217787af, in groundhog_system::GroundhogSystem::initDeviceContainer()
[ros2_control_node-3] #11   Object "/opt/ros/jazzy/lib/libdevice_container.so", at 0x7f90400bdcdd, in ros2_canopen::DeviceContainer::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-3] #10   Object "/opt/ros/jazzy/lib/libdevice_container.so", at 0x7f90400bfdc1, in ros2_canopen::DeviceContainer::load_drivers()
[ros2_control_node-3] #9    Object "/ros_ws/install/roboteq_driver/lib/libnode_canopen_roboteq_driver.so", at 0x7f90215cf544, in ros2_canopen::node_interfaces::NodeCanopenDriver<rclcpp::Node>::activate()
[ros2_control_node-3] #8    Object "/ros_ws/install/roboteq_driver/lib/libnode_canopen_roboteq_driver.so", at 0x7f9021549e9c, in ros2_canopen::node_interfaces::NodeCanopenRoboteqDriver<rclcpp::Node>::add_to_master()
[ros2_control_node-3] #7    Object "/opt/ros/jazzy/lib/libnode_canopen_base_driver.so", at 0x7f90210cd7bb, in 
[ros2_control_node-3] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x7f905bb9d390, in __cxa_throw
[ros2_control_node-3] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x7f905bb87a54, in std::terminate()
[ros2_control_node-3] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x7f905bb9d0d9, in 
[ros2_control_node-3] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x7f905bb87ff4, in 
[ros2_control_node-3] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f905b8ca8fe]
[ros2_control_node-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f905b8e727d]
[ros2_control_node-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f905b940b2c]
```
**Setup:**
 - Device: Motor controller - Roboteq 
 - OS: Ubuntu 24.04
 - ROS-Distro: Jazzy
 - Branch/Commit: Latest master

ipa-vsp, 2025-08-15:
Hi @jclinton830, have you added `boot_timeout_ms: 1000` in *bus.yml*? You might also consider increasing it, as the default value may be too low. The default value is set to 20ms. Maybe it needs to be increased.

closed 2025-08-15 by ipa-vsp

jclinton830, 2025-08-16:
@ipa-vsp I can certainly try your new PR; however, at the time we did try boot_time_ms as 20000 and still had trouble. I can try again and let you know the outcome.

jclinton830, 2025-08-25:
I can confirm that after adding boot_timeout_ms to all the slave nodes, it works with the official binaries.